elktownlabs/murgcam #26: Modem scheint nicht richtig zu antworten

sirtet, 2022-11-12:
```
I (12844) main: Uploading picture to server
I (12854) main: Power down modem for 2s
I (14854) main: Power up modem
I (15354) main: Power key on for 1.2s
I (16554) main: Power key off and wait 1s
I (17554) main: Init modem
I (17554) cellmodem: Init DTE
I (17554) uart: queue free spaces: 30
I (17554) cellmodem: Init SIM800
I (17554) esp-modem: Sending command: AT
E (18054) esp-modem: esp_modem_dte_send_cmd(231): process command timeout
E (18054) dce_service: esp_modem_dce_sync(48): send command failed
E (18054) sim800: sim800_init(656): sync failed
I (18054) cellmodem: Modem not yet initialized.
I (19064) esp-modem: Sending command: AT
I (19064) esp-modem: Line received: ▒▒AT
 (ff ff)
I (19114) esp-modem: Line received: OK
 (4f 4b)
I (19114) esp-modem: Sending command: ATE0
I (19164) esp-modem: Line received: ATE0
 (41 54)
E (19164) esp-modem: esp_dte_handle_line(92): handle line failed
W (19164) cellmodem: Unknown line received: ATE0
 (41 54 45 30 d d)
I (19214) esp-modem: Line received: OK
 (4f 4b)
I (19214) esp-modem: Sending command: AT+CGMM
I (19264) esp-modem: Line received:
 (d a)
I (19314) esp-modem: Line received: SIMCOM_SIM800C
 (53 49)
I (19364) esp-modem: Line received:
 (d a)
I (19414) esp-modem: Line received: OK
 (4f 4b)
I (19414) esp-modem: Sending command: AT+CGSN
I (19464) esp-modem: Line received:
 (d a)
I (19514) esp-modem: Line received: +CPIN: READY
 (2b 43)
I (19564) esp-modem: Line received:
 (d a)
I (19614) esp-modem: Line received: 865532044865174
 (38 36)
I (19664) esp-modem: Line received:
 (d a)
I (19714) esp-modem: Line received: OK
 (4f 4b)
I (19714) esp-modem: Sending command: AT+CPIN?
I (19764) esp-modem: Line received:
 (d a)
I (19814) esp-modem: Line received: +CPIN: READY
 (2b 43)
I (19864) esp-modem: Line received:
 (d a)
I (19914) esp-modem: Line received: OK
 (4f 4b)
I (19914) esp-modem: Sending command: AT+CREG?
I (19964) esp-modem: Line received:
 (d a)
I (20014) esp-modem: Line received: +CREG: 0,2
 (2b 43)
I (20064) esp-modem: Line received:
 (d a)
I (20114) esp-modem: Line received: OK
 (4f 4b)
E (20114) sim800: sim800_wait_registration(524): waiting for network registration failed
E (20114) sim800: sim800_init(670): waiting for registration timed out
I (20124) cellmodem: Modem not yet initialized.
I (21124) esp-modem: Sending command: AT
I (21124) esp-modem: Line received:
 (d a)
I (21174) esp-modem: Line received: OK
 (4f 4b)
I (21174) esp-modem: Sending command: ATE0
I (21224) esp-modem: Line received:
 (d a)
I (21274) esp-modem: Line received: OK
 (4f 4b)
I (21274) esp-modem: Sending command: AT+CGMM
I (21324) esp-modem: Line received:
 (d a)
I (21374) esp-modem: Line received: SIMCOM_SIM800C
 (53 49)
I (21424) esp-modem: Line received:
 (d a)
I (21474) esp-modem: Line received: OK
 (4f 4b)
I (21474) esp-modem: Sending command: AT+CGSN
I (21524) esp-modem: Line received:
 (d a)
I (21574) esp-modem: Line received: 865532044865174
 (38 36)
I (21624) esp-modem: Line received:
 (d a)
I (21674) esp-modem: Line received: OK
 (4f 4b)
I (21674) esp-modem: Sending command: AT+CPIN?
I (21724) esp-modem: Line received:
 (d a)
I (21774) esp-modem: Line received: +CPIN: READY
 (2b 43)
I (21824) esp-modem: Line received:
 (d a)
I (21874) esp-modem: Line received: OK
 (4f 4b)
I (21874) esp-modem: Sending command: AT+CREG?
I (21924) esp-modem: Line received:
 (d a)
I (21974) esp-modem: Line received: +CREG: 0,2
 (2b 43)
I (22024) esp-modem: Line received:
 (d a)
I (22074) esp-modem: Line received: OK
 (4f 4b)
E (22074) sim800: sim800_wait_registration(524): waiting for network registration failed
E (22074) sim800: sim800_init(670): waiting for registration timed out
I (22084) cellmodem: Modem not yet initialized.
I (22494) esp-modem: Line received:
 (d a)
I (22544) esp-modem: Line received: Call Ready
 (43 61)
E (22544) esp-modem: esp_dte_handle_line(91): no handler for line
W (22544) cellmodem: Unknown line received: Call Ready
 (43 61 6c 6c 20 52)
I (23084) esp-modem: Sending command: AT
I (23084) esp-modem: Line received:
 (d a)
I (23134) esp-modem: Line received: OK
 (4f 4b)
I (23134) esp-modem: Sending command: ATE0
I (23184) esp-modem: Line received:
 (d a)
I (23234) esp-modem: Line received: OK
 (4f 4b)
I (23234) esp-modem: Sending command: AT+CGMM
I (23284) esp-modem: Line received:
 (d a)
I (23334) esp-modem: Line received: SIMCOM_SIM800C
 (53 49)
I (23384) esp-modem: Line received:
 (d a)
I (23434) esp-modem: Line received: OK
 (4f 4b)
I (23434) esp-modem: Sending command: AT+CGSN
I (23484) esp-modem: Line received:
 (d a)
I (23534) esp-modem: Line received: SMS Ready
 (53 4d)
I (23584) esp-modem: Line received:
 (d a)
I (23634) esp-modem: Line received: 865532044865174
 (38 36)
I (23684) esp-modem: Line received:
 (d a)
I (23734) esp-modem: Line received: OK
 (4f 4b)
I (23734) esp-modem: Sending command: AT+CPIN?
I (23784) esp-modem: Line received:
 (d a)
I (23834) esp-modem: Line received: +CPIN: READY
 (2b 43)
I (23884) esp-modem: Line received:
 (d a)
I (23934) esp-modem: Line received: OK
 (4f 4b)
I (23934) esp-modem: Sending command: AT+CREG?
I (23984) esp-modem: Line received:
 (d a)
I (24034) esp-modem: Line received: +CREG: 0,3
 (2b 43)
I (24084) esp-modem: Line received:
 (d a)
I (24134) esp-modem: Line received: OK
 (4f 4b)
E (24134) sim800: sim800_wait_registration(524): waiting for network registration failed
E (24134) sim800: sim800_init(670): waiting for registration timed out
I (24144) cellmodem: Modem not yet initialized.
I (25144) esp-modem: Sending command: AT
I (25144) esp-modem: Line received:
 (d a)
I (25194) esp-modem: Line received: OK
 (4f 4b)
I (25194) esp-modem: Sending command: ATE0
I (25244) esp-modem: Line received:
 (d a)
I (25294) esp-modem: Line received: OK
 (4f 4b)
I (25294) esp-modem: Sending command: AT+CGMM
I (25344) esp-modem: Line received:
 (d a)
I (25394) esp-modem: Line received: SIMCOM_SIM800C
 (53 49)
I (25444) esp-modem: Line received:
 (d a)
I (25494) esp-modem: Line received: OK
 (4f 4b)
I (25494) esp-modem: Sending command: AT+CGSN
I (25544) esp-modem: Line received:
 (d a)
I (25594) esp-modem: Line received: 865532044865174
 (38 36)
I (25644) esp-modem: Line received:
 (d a)
I (25694) esp-modem: Line received: OK
 (4f 4b)
I (25694) esp-modem: Sending command: AT+CPIN?
I (25744) esp-modem: Line received:
 (d a)
I (25794) esp-modem: Line received: +CPIN: READY
 (2b 43)
I (25844) esp-modem: Line received:
 (d a)
I (25894) esp-modem: Line received: OK
 (4f 4b)
I (25894) esp-modem: Sending command: AT+CREG?
I (25944) esp-modem: Line received:
 (d a)
I (25994) esp-modem: Line received: +CREG: 0,3
 (2b 43)
I (26044) esp-modem: Line received:
 (d a)
I (26094) esp-modem: Line received: OK
 (4f 4b)
E (26094) sim800: sim800_wait_registration(524): waiting for network registration failed
E (26094) sim800: sim800_init(670): waiting for registration timed out
```

elktownlabs, 2022-11-13:
Das Modem antwortet.
Siehe [https://m2msupport.net/m2msupport/atcreg-network-registration/](https://m2msupport.net/m2msupport/atcreg-network-registration/)

`I (25994) esp-modem: Line received: +CREG: 0,3`
Er sucht hier nach einem GSM Netz und findet noch keines. Evtl. ne Weile laufen lassen. Das erste mal dauert es immer relativ lange. Irgendwann bricht er ab falls es garnicht geht. Und auch unbedingt eine Antenne anschließen. Offener SMA-Anschluss ist nicht unbedingt so Ideal wegen Reflektionen, Störungen der restlichen Elektronik usw.

sirtet, 2022-11-18:
Ja, bricht nach paar Min. ab. Antenne hab ich die WIFI dran, wohl nicht ideal.
Muss morgen noch andere SIM probieren, hab noch keine getestet von der ich sicher bin dass sie hier auf GSM läuft, habe aber eine solche von 1nce.com
```
E (207894) sim800: sim800_wait_registration(524): waiting for network registration failed
E (207894) sim800: sim800_init(670): waiting for registration timed out
I (207904) cellmodem: Modem not yet initialized.
I (208904) esp-modem: Sending command: AT
I (208904) esp-modem: Line received:
 (d a)
I (208954) esp-modem: Line received: OK
 (4f 4b)
I (208954) esp-modem: Sending command: ATE0
I (209004) esp-modem: Line received:
 (d a)
I (209054) esp-modem: Line received: OK
 (4f 4b)
I (209054) esp-modem: Sending command: AT+CGMM
I (209104) esp-modem: Line received:
 (d a)
I (209154) esp-modem: Line received: SIMCOM_SIM800C
 (53 49)
I (209204) esp-modem: Line received:
 (d a)
I (209254) esp-modem: Line received: OK
 (4f 4b)
I (209254) esp-modem: Sending command: AT+CGSN
I (209304) esp-modem: Line received:
 (d a)
I (209354) esp-modem: Line received: 865532044865174
 (38 36)
I (209404) esp-modem: Line received:
 (d a)
I (209454) esp-modem: Line received: OK
 (4f 4b)
I (209454) esp-modem: Sending command: AT+CPIN?
I (209504) esp-modem: Line received:
 (d a)
I (209554) esp-modem: Line received: +CPIN: READY
 (2b 43)
I (209604) esp-modem: Line received:
 (d a)
I (209654) esp-modem: Line received: OK
 (4f 4b)
I (209654) esp-modem: Sending command: AT+CREG?
I (209704) esp-modem: Line received:
 (d a)
I (209754) esp-modem: Line received: +CREG: 0,5
 (2b 43)
I (209804) esp-modem: Line received:
 (d a)
I (209854) esp-modem: Line received: OK
 (4f 4b)
E (209854) sim800: sim800_wait_registration(524): waiting for network registration failed
E (209854) sim800: sim800_init(670): waiting for registration timed out
I (209864) cellmodem: Modem not yet initialized.
I (210864) esp-modem: Sending command: AT
I (210864) esp-modem: Line received:
 (d a)
I (210914) esp-modem: Line received: OK
 (4f 4b)
I (210914) esp-modem: Sending command: ATE0
I (210964) esp-modem: Line received:
 (d a)
I (211014) esp-modem: Line received: OK
 (4f 4b)
I (211014) esp-modem: Sending command: AT+CGMM
I (211064) esp-modem: Line received:
 (d a)
I (211114) esp-modem: Line received: SIMCOM_SIM800C
 (53 49)
I (211164) esp-modem: Line received:
 (d a)
I (211214) esp-modem: Line received: OK
 (4f 4b)
I (211214) esp-modem: Sending command: AT+CGSN
I (211264) esp-modem: Line received:
 (d a)
I (211314) esp-modem: Line received: 865532044865174
 (38 36)
I (211364) esp-modem: Line received:
 (d a)
I (211414) esp-modem: Line received: OK
 (4f 4b)
I (211414) esp-modem: Sending command: AT+CPIN?
I (211464) esp-modem: Line received:
 (d a)
I (211514) esp-modem: Line received: +CPIN: READY
 (2b 43)
I (211564) esp-modem: Line received:
 (d a)
I (211614) esp-modem: Line received: OK
 (4f 4b)
I (211614) esp-modem: Sending command: AT+CREG?
I (211664) esp-modem: Line received:
 (d a)
I (211714) esp-modem: Line received: +CREG: 0,5
 (2b 43)
I (211764) esp-modem: Line received:
 (d a)
I (211814) esp-modem: Line received: OK
 (4f 4b)
E (211814) sim800: sim800_wait_registration(524): waiting for network registration failed
E (211814) sim800: sim800_init(670): waiting for registration timed out
I (211824) cellmodem: Modem not yet initialized.
I (212824) cellmodem: Modem initialization failed. Not uploading.
I (212824) main: Closing modem connection
I (212824) main: Modem power key on for 1.2s
I (214024) main: Modem power key off and wait 1s
I (214344) esp-modem: Line received:
 (d a)
I (214394) esp-modem: Line received: NORMAL POWER DOWN
 (4e 4f)
E (214394) esp-modem: esp_dte_handle_line(91): no handler for line
W (214394) cellmodem: Unknown line received: NORMAL POWER DOWN
 (4e 4f 52 4d 41 4c)
W (214814) esp-modem: Rx Break
I (215024) main: Power down modem now
I (215024) main: Using sleep time from config: 1800 secs
I (215024) main: Up for 213710 ms. Going to sleep for 1586290 ms
```

sirtet, 2022-11-19:
Die GSM Antenne und SIM vom RiverSpy bringen keine Veränderung. 
Der Riverspy, der auch ein SIM800 Modul verwendet, bekommt eine Verbindung, also sind Antenne und SIM ok.

sirtet, 2022-11-19:
Korrektur, bringt doch etwas. __Modem registriert sich!__ 
Aber... es dauert der cam zu lange?
Warum sagt sie `waiting for network registration failed`?
```
...
I (211614) esp-modem: Sending command: AT+CREG?
I (211664) esp-modem: Line received:
 (d a)
I (211714) esp-modem: Line received: +CREG: 0,5
 (2b 43)
I (211764) esp-modem: Line received:
 (d a)
I (211814) esp-modem: Line received: OK
 (4f 4b)
E (211814) sim800: sim800_wait_registration(524): waiting for network registration failed
E (211814) sim800: sim800_init(670): waiting for registration timed out
```
Mit der ersten SIM wurde ja eine Weile `0,2`, dann `+CREG: 0,3` gemeldet, laut obigem Link _registration denied_...

jetzt haben wir nach einer Weile `0,5`, also _registered, roaming_
Was läuft schief?

Dump von "kameraaufnahme nach boot"  bis "registriert":
```
I (8204) camhandler: Waiting for camera to settle
I (12364) main: Shortened data from 76800 to 75604.
I (12364) main: Picture taken. Size = 75604
I (12364) main: Shutting down and switching off camera
I (12374) main: Uploading picture to server
I (12374) main: Power down modem for 2s
I (14384) main: Power up modem
I (14884) main: Power key on for 1.2s
I (16084) main: Power key off and wait 1s
I (17084) main: Init modem
I (17084) cellmodem: Init DTE
I (17084) uart: queue free spaces: 30
I (17084) cellmodem: Init SIM800
I (17084) esp-modem: Sending command: AT
E (17584) esp-modem: esp_modem_dte_send_cmd(231): process command timeout
E (17584) dce_service: esp_modem_dce_sync(48): send command failed
E (17584) sim800: sim800_init(656): sync failed
I (17584) cellmodem: Modem not yet initialized.
I (18594) esp-modem: Sending command: AT
I (18594) esp-modem: Line received: ▒▒AT
 (ff ff)
I (18644) esp-modem: Line received: OK
 (4f 4b)
I (18644) esp-modem: Sending command: ATE0
I (18694) esp-modem: Line received: ATE0
 (41 54)
E (18694) esp-modem: esp_dte_handle_line(92): handle line failed
W (18694) cellmodem: Unknown line received: ATE0
 (41 54 45 30 d d)
I (18744) esp-modem: Line received: OK
 (4f 4b)
I (18744) esp-modem: Sending command: AT+CGMM
I (18794) esp-modem: Line received:
 (d a)
I (18844) esp-modem: Line received: SIMCOM_SIM800C
 (53 49)
I (18894) esp-modem: Line received:
 (d a)
I (18944) esp-modem: Line received: OK
 (4f 4b)
I (18944) esp-modem: Sending command: AT+CGSN
I (18994) esp-modem: Line received:
 (d a)
I (19044) esp-modem: Line received: 865532044865174
 (38 36)
I (19094) esp-modem: Line received:
[... 262 lines cut ...]
I (29084) esp-modem: Line received:
 (d a)
I (29134) esp-modem: Line received: OK
 (4f 4b)
I (29134) esp-modem: Sending command: AT+CPIN?
I (29184) esp-modem: Line received:
 (d a)
I (29234) esp-modem: Line received: +CPIN: READY
 (2b 43)
I (29284) esp-modem: Line received:
 (d a)
I (29334) esp-modem: Line received: OK
 (4f 4b)
I (29334) esp-modem: Sending command: AT+CREG?
I (29384) esp-modem: Line received:
 (d a)
I (29434) esp-modem: Line received: +CREG: 0,5
 (2b 43)
I (29484) esp-modem: Line received:
 (d a)
I (29534) esp-modem: Line received: OK
 (4f 4b)
E (29534) sim800: sim800_wait_registration(524): waiting for network registration failed
E (29534) sim800: sim800_init(670): waiting for registration timed out
```

elktownlabs, 2022-11-27:
Versuche mal folgende Änderung. Damit sollte die Kamera auch ein Roaming-Netz verwenden:

```
--- a/firmware/elkcam_firmware/components/modem/src/sim800.c
+++ b/firmware/elkcam_firmware/components/modem/src/sim800.c
@@ -211,7 +211,7 @@ static esp_err_t sim800_handle_waitregistration(modem_dce_t *dce, const char *li
         err = esp_modem_process_command_done(dce, MODEM_STATE_FAIL);
     } else if (!strncmp(line, "+CREG: ", strlen("+CREG: "))) {
         int value = line[9] - '0';
-        if (value ==1) {
+        if ((value ==1) || (value == 5)) {
             dce->registered = 1;
         }
         err = ESP_OK;
@@ -693,4 +693,4 @@ esp_err_t sim800_get_engineering_data(sim800_engineering_data_t* data)
         }
     }
     return res;
-}
\ No newline at end of file
+}
```

sirtet, 2022-11-27:
Stimmt genau... wenn du das schon als diff hast, kannst du das committen?

Was ist mit der EOF Änderung?

elktownlabs, 2022-12-01:
EOF-Newline hat wahrscheinlich mein Editor automatisch weggemacht. Ist irrelevant.
Ich mache einen Commit aber als bedingt reinkompiliert. Dann kann man es bei Bedarf noch rausnehmen.