elktownlabs/murgcam #27: Kamera Probleme

sirtet, 2022-11-12:
Kamera hat Per Wifi Config UI 3-4 (schlechte) Bilder aufgenommen, jetzt keine mehr...
Erste 2 Bilder waren grünstichig und "verschoben"...
Hmm, ist das Linsen Cover grün? Wegnehmen...
Nächste Bilder sind nur grosse Farb-Pixel, Bunt wie von TV Testbild...
Dann plötzlich keine Bilder mehr, aber Fehler in der Konsole:

```
I (25824) OV2640: Set 146 to 1
I (25824) OV2640: Set 147 to 228
I (25824) camhandler: Setting sharpness to 0
I (25834) camhandler: Setting flip/mirror to 0
I (25834) camhandler: Waiting for camera to settle
I (29994) httpserver: Picture taken. Size = 45056
W (30024) httpd_txrx: httpd_sock_err: error in send : 104
I (30024) httpserver: Request headers lost
W (30024) httpd_txrx: httpd_sock_err: error in recv : 104
```

elktownlabs, 2022-11-13:
Da ist irgendetwas mit dem Datenstrom von dem Kameramodul nicht ok. Normalerweise müsste noch eine Meldung `Shortened data from ... to ...` nach dem `Picture taken. Size=...` kommen. 

Das könnte bedeuten es kommt kein gültiges JPEG Bild an. Ich habe heute meine Murg-Cam zur Wartung heimgeholt und poste mal die Bild-Einstellungen sobald ich sie am Strom habe. 

Das Kameramodul reagiert teilweise sehr energisch auf manche nicht vorgesehenen Konfigurationseinstellungen

sirtet, 2022-11-13:
Ja, erfolgreiche Bilder waren vor ich Parameter geändert habe...
Habe jetzt Defaults aus Code übernommen, hat nicht geholfen.
https://github.com/elktownlabs/murgcam/blob/98d05d2cdfb141ffb79fd98262dab560fdc96833/firmware/elkcam_firmware/main/config.h#L19-L28

sirtet, 2022-11-17:
Weitere Beobachtungen:

Typische Meldungen:
```
I (440814) httpserver: Host 192.168.4.1 requested a picture.
I (440814) httpserver: Taking picture
I (440814) camhandler: Reinitializing cam
I (441114) OV2640: HID:9794
I (441154) camhandler: Configuring camera to global config settings
I (441154) camhandler: Setting pixel timing to 0
I (441154) camhandler: Setting picture quality to 63
I (441164) camhandler: Setting auto exposure to 2
I (441164) camhandler: Setting light mode to 0
I (441174) OV2640: Set 255 to 0
I (441174) OV2640: Set 124 to 0
I (441184) OV2640: Set 125 to 0
I (441184) camhandler: Setting color saturation to 0
I (441194) OV2640: Set 255 to 0
I (441194) OV2640: Set 124 to 0
I (441194) OV2640: Set 125 to 0
I (441204) camhandler: Setting brightness to 0
I (441204) OV2640: Set 255 to 0
I (441214) OV2640: Set 124 to 0
I (441214) OV2640: Set 125 to 0
I (441214) camhandler: Setting contrast to 0
I (441224) OV2640: Set 255 to 0
I (441224) OV2640: Set 124 to 0
I (441234) OV2640: Set 125 to 0
I (441234) camhandler: Setting hue to 0
I (441244) OV2640: Set 255 to 0
I (441244) OV2640: Set 146 to 1
I (441244) OV2640: Set 147 to 228
I (441254) camhandler: Setting sharpness to 0
I (441254) camhandler: Setting flip/mirror to 0
I (441264) camhandler: Waiting for camera to settle
I (445414) httpserver: Picture taken. Size = 40960
I (445514) httpserver: Request headers lost
```
- Quality scheint rückwärz zu funktionieren, Q=63 gibt kleinste Grösse, um 40k (`Picture taken. Size = 40960`). Q=10 ergibt um 80k.

- Ein Bild wird scheinbar schon aufgenommen, Linse abdecken gibt kleinere Grösse als gut beleuchtet in die Kamera schauen, allerdings nur um 10%

- Schaut die Antwort vom Sensor `OV2640: Set 255 to 0` etc.  gesund aus? Oder kann man daraus etwas ablesen?
Ist immer gleich, ausser bei Hue...
- ` httpserver: Request headers lost` ist normal?
- `httpd_sock_err: error in send : 104` scheint mit dem Client zu tun zu haben, sehe ich nur in Chrome, nicht in Firefox.
- Einmaliges Vorkommnis: Nach dem Ändern von Cell Pin von NULL-String auf 6400 stürzte die Kamera/(das System?) bis zum nächsten Reset ab, jede Bildanforderung sah so aus:
```
I (2297984) httpserver: Host 192.168.4.1 requested a picture.
I (2297984) httpserver: Taking picture
I (2297984) camhandler: Reinitializing cam
E (2298194) i2c: i2c command link malloc error
E (2298194) i2c: i2c command link malloc error
E (2298204) i2c: i2c command link malloc error
E (2298204) i2c: i2c command link malloc error
E (2298314) i2c: i2c command link malloc error
E (2298324) i2c: i2c command link malloc error
E (2298334) i2c: i2c command link malloc error
E (2298344) i2c: i2c command link malloc error
E (2298354) i2c: i2c command link malloc error
E (2298364) i2c: i2c command link malloc error
E (2298374) i2c: i2c command link malloc error
E (2298374) i2c: i2c command link malloc error
I (2298374) OV2640: HID:9794
E (2298374) i2c: i2c command link malloc error
E (2298374) i2c: i2c command link malloc error
E (2298394) i2c: i2c command link malloc error
E (2298394) i2c: i2c command link malloc error
E (2298394) i2c: i2c command link malloc error
E (2298394) i2c: i2c command link malloc error
[... 502 lines cut ...]
E (2300984) i2c: i2c command link malloc error
I (2300994) OV2640: Set 125 to 30
I (2300994) camhandler: Setting hue to 0
E (2300994) i2c: i2c command link malloc error
E (2300994) i2c: i2c command link malloc error
I (2301004) OV2640: Set 255 to 0
E (2301004) i2c: i2c command link malloc error
E (2301004) i2c: i2c command link malloc error
I (2301024) OV2640: Set 146 to 1
E (2301024) i2c: i2c command link malloc error
E (2301034) i2c: i2c command link malloc error
E (2301034) i2c: i2c command link malloc error
E (2301034) i2c: i2c command link malloc error
I (2301044) OV2640: Set 147 to 32
I (2301044) camhandler: Setting sharpness to 0
E (2301044) i2c: i2c command link malloc error
E (2301054) i2c: i2c command link malloc error
E (2301064) i2c: i2c command link malloc error
E (2301074) i2c: i2c command link malloc error
E (2301084) i2c: i2c command link malloc error
E (2301084) i2c: i2c command link malloc error
I (2301094) camhandler: Setting flip/mirror to 0
I (2301094) camhandler: Waiting for camera to settle
E (2313094) httpserver: Camera timed out
```

sirtet, 2022-11-18:
Ach, schlechte Lötstelle... 
An Kamera Anschluss J4. Pin18 = CSI_VSYNC war nicht verbunden...
Bild aufnehmen über Wifi läuft.

closed 2022-11-18 by sirtet